facebookresearch/habitat-lab #2109: Depth Image Definition

realjoshqsun, 2024-11-21:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.1.7 

Habitat-Sim: v0.1.7 

Hello,

I'm wondering what the depth observations mean? I printed out the output matrix from depth observation, and it's within the range of [0, 1]. The min and max are not always 0 and 1. I also noticed that there's some noise in it.

I'd like to know that is it representing some kind of relative distance like depthanything, or it's simulating a particular type of depth sensor. 

I read through the Docs in https://aihabitat.org/docs/habitat-lab/, but I can't find a answer.

realjoshqsun, 2024-11-27:
This has been resolved.

closed 2024-11-27 by realjoshqsun

LeungChaos, 2025-03-24:
So may I ask what‘s the meaning of the depth sensor values? Why are they all numbers less than 1? Are they in meters?