facebookresearch/habitat-lab #2115: Dose the NVIVIA A6000 support the use of habitat-lab?

zhai-create, 2024-12-05:
When I use the NVIVIA 6000, I met the issue when the habitat is initialized:
GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValue
Aborted (core dumped)
What should I do to solve this problem?
[image]

zhai-create, 2024-12-05:
I am using the NVIDIA A6000 graphics card.

0mdc, 2024-12-11:
@zhai-create Could you provide some more information on your setup?

1. What OS are you using?
2. Could you provide the output of `conda list`?
3. How did you install `habitat-sim`? Did you use the conda package, or did you compile it yourself?

zhai-create, 2024-12-30:
> @zhai-create Could you provide some more information on your setup?
> 
> 1. What OS are you using?
> 2. Could you provide the output of `conda list`?
> 3. How did you install `habitat-sim`? Did you use the conda package, or did you compile it yourself?

I have solved this problem by using docker. Could please help me to solve the problem of https://github.com/facebookresearch/habitat-lab/issues/2125?

zundongke, 2025-01-07:
I have met the same problem. Could you give the solution to solve it ?

0mdc, 2025-01-13:
@zundongke : Can you try installing the latest habitat-sim nightly? There were conda dependencies that caused this issue.

See: https://github.com/facebookresearch/habitat-sim/pull/2519

Lazy-boy193, 2025-02-28:
> When I use the NVIVIA 6000, I met the issue when the habitat is initialized:当我使用 NVIVIA 6000 时，我在初始化 habitat 时遇到了这个问题： GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValueGL::Context: 无法检索 OpenGL 版本：GL::Renderer::Error::InvalidValue Aborted (core dumped)  已中止（核心转储） What should I do to solve this problem?我应该怎么做才能解决这个问题？ [image]

I encountered the same issue. How did you resolve it?

Eku127, 2025-04-03:
> [@zundongke](https://github.com/zundongke) : Can you try installing the latest habitat-sim nightly? There were conda dependencies that caused this issue.
> 
> See: [facebookresearch/habitat-sim#2519](https://github.com/facebookresearch/habitat-sim/pull/2519)

Any solutions to this problem? As I compile the habitat-sim by myself, issue #2561

shayan223, 2025-05-06:
I'm unsure if folks are still running into this issue, but I was able to solve it for myself by doing a clean install with a new conda environment, and more specifically, noting that the README instructions for habitat-lab uses a different repository for their stable release:

git clone --branch stable https://github.com/facebookresearch/habitat-lab.git
cd habitat-lab
pip install -e habitat-lab  # install habitat_lab

Not using the stable branch after glossing over that instruction was the root cause of most my issues.

yourkln, 2025-11-06:
I had exactly the same problem and managed to solve it. The issue is that the system is trying to use Mesa's software rendering libraries instead of the NVIDIA GPU libraries. You should replace the Mesa EGL libraries with the NVIDIA ones by creating symlinks. If you need more details, let me know.

ChengLiDuoJi, 2025-12-25:
> 我遇到了完全一样的问题，也成功解决了。问题在于系统试图使用Mesa的软件渲染库，而不是NVIDIA GPU的库。你应该用NVIDIA的库替换Mesa EGL库，创建符号链接。如果你需要更多细节，告诉我。

I need details!!! thanx

ChengLiDuoJi, 2025-12-25:
> > 我遇到了完全一样的问题，也成功解决了。问题在于系统试图使用Mesa的软件渲染库，而不是NVIDIA GPU的库。你应该用NVIDIA的库替换Mesa EGL库，创建符号链接。如果你需要更多细节，告诉我。
> 
> 是的，我也遇到了同样的问题，我需要更多细节，你是怎么解决的？作系统：Ubuntu 20.24，我用的是NVIDIA RTX 4070。

Have you sovled this problem? How did you solve?