facebookresearch/habitat-lab #2116: How to Change Humanoid Appearance or Config Using env.reset()?

wsakobe, 2024-12-09:
Hi! I am currently using Habitat 3.0 to establish a social navigation task that involves multiple humanoid agents in the environment. I have generated several episodes, where each episode specifies the corresponding URDF and motion data files for each humanoid agent based on their IDs.

My goal is to utilize `env.reset()` to automatically switch to the next episode, updating the configurations for each humanoid agent (e.g., appearance and motion files) dynamically, similar to how other tasks operate. To achieve this, I attempted to update these configurations using `habitat.get_config()` and invoked `sim.reconfigure()` within the simulator to apply the changes. However, this approach did not succeed, as the humanoid agents’ appearances remained unchanged across episodes.

I suspect that the issue lies in my implementation of the `sim.reconfigure()` function, which is largely based on the settings from `RearrangeSim`.

Could you kindly advise on how to dynamically update these humanoid configurations during episode transitions? I would greatly appreciate any guidance or suggestions you can provide.

Thank you for your time and assistance.

0mdc, 2024-12-11:
Hi @wsakobe!

You may want to check out this early implementation: https://github.com/facebookresearch/habitat-lab/blob/9783e3d744175adc6bc93a58e8c182d2b83948a9/habitat-hitl/habitat_hitl/environment/avatar_switcher.py
The code is not yet factored in, but it should do the trick for handling your use case.

wsakobe, 2025-01-21:
@0mdc Thank you for your valuable response. I have implemented the required functionality by referring to `avatar_switcher.py`.

closed 2025-01-21 by wsakobe

wsakobe, 2025-02-10:
@0mdc Sorry to bother you again. I recently discovered that switching the humanoid avatar in `env.reset()` still has issues. Specifically, I perform the avatar switch in `self.task.reset()`, but when resetting at a random episode, calling `sim.get_articulated_object_manager().remove_object_by_handle` results in a segmentation fault (core dumped) error. Below is my switch_avatar function. Could you please take a look and see if there are any issues in my implementation?

```
def _switch_avatar(self, sim, episode: Episode) -> None:
        main_name_attr = "main_humanoid_name"
        main_humanoid_name = episode.info[main_name_attr]
        extra_name_attr = "extra_humanoid_names"
        extra_humanoid_names = episode.info[extra_name_attr]

        humanoid_agent_data: ArticulatedAgentData = sim.agents_mgr[0]
        humanoid = humanoid_agent_data.articulated_agent
        sim.get_articulated_object_manager().remove_object_by_handle(
            humanoid.sim_obj.handle
        )
        
        for i in range(len(extra_humanoid_names)):
            humanoid_agent_data: ArticulatedAgentData = sim.agents_mgr[i+2]
            humanoid = humanoid_agent_data.articulated_agent
            
            sim.get_articulated_object_manager().remove_object_by_handle(
                humanoid.sim_obj.handle
            )

        # Main human avatar switch
        urdf_path = "data/humanoids/humanoid_data/" + main_humanoid_name + "/" + main_humanoid_name + ".urdf"
        humanoid_motion_data_path = "data/humanoids/humanoid_data/" + main_humanoid_name + "/" + main_humanoid_name + "_motion_data_smplx.pkl"
        humanoid_agent_data: ArticulatedAgentData = sim.agents_mgr[0]
        agent_config = DictConfig(
            {
                "articulated_agent_urdf": urdf_path,
                "motion_data_path": humanoid_motion_data_path,
            }
        )
        humanoid = KinematicHumanoid(agent_config, sim=sim)
        humanoid.reconfigure()
        humanoid_agent_data.articulated_agent = humanoid

        # other human avatar switch
        for i, extra_human_name in enumerate(extra_humanoid_names):
            urdf_path = "data/humanoids/humanoid_data/" + extra_human_name + "/" + extra_human_name + ".urdf"
            humanoid_motion_data_path = "data/humanoids/humanoid_data/" + extra_human_name + "/" + extra_human_name + "_motion_data_smplx.pkl"
            humanoid_agent_data: ArticulatedAgentData = sim.agents_mgr[i+2]
            agent_config = DictConfig(
                {
                    "articulated_agent_urdf": urdf_path,
                    "motion_data_path": humanoid_motion_data_path,
                }
            )
            humanoid = KinematicHumanoid(agent_config, sim=sim)
            humanoid.reconfigure()
            humanoid_agent_data.articulated_agent = humanoid
```

wsakobe, 2025-02-10:
Additionally, I have noticed that this error often occurs after approximately 72 calls to `remove_object_by_handle()`. I’m not sure if this information can help you identify the issue.