facebookresearch/habitat-lab #2142: X Error of failed request:  BadAccess (attempt to access private resource denied)

BJHYZJ, 2025-01-26:
I still meet this problem in ubuntu22.04 with RTX3090ti. I don't found great solution. Is this problem still not solved effectively? Because in the README.md of V0.3.2, the description of this problem still exists?

RookieXwc, 2025-02-18:
I still encounter this issue on Ubuntu 22.04 and RTX 2080Ti, and the habitat lab version is v0.3.3. So since Ubuntu 20.04 and 22.04 both have this issue, do we need to use 18.04? @aclegg3 Thank you very much for your help!

ChungHaLee, 2025-02-24:
Just encountered same issue here

vlamirand, 2025-03-12:
(habitat) ➜  habitat-lab git:(master) ✗ /home/o/miniconda3/envs/habitat/bin/python /home/o/Documents/habitat-lab/examples/interactive_play.py     
pybullet build time: Mar 10 2025 20:07:54
PluginManager::Manager: duplicate static plugin StbImageImporter, ignoring
PluginManager::Manager: duplicate static plugin GltfImporter, ignoring
PluginManager::Manager: duplicate static plugin BasisImporter, ignoring
PluginManager::Manager: duplicate static plugin AssimpImporter, ignoring
PluginManager::Manager: duplicate static plugin AnySceneImporter, ignoring
PluginManager::Manager: duplicate static plugin AnyImageImporter, ignoring
2025-03-12 10:30:03,681 Initializing dataset RearrangeDataset-v0
2025-03-12 10:30:19,226 initializing sim RearrangeSim-v0
[10:30:19:251416]:[Warning]:[Metadata] AbstractAttributesManager.h(531)::buildAttrSrcPathsFromJSONAndLoad : <Articulated Object> : No Glob path result found for `data/hab_fetch_1.0/robots/fetch_no_base.urdf` so unable to load templates from that path.
[10:30:19:299567]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_00` Value : `navmeshes/v3_sc4_staging_00.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299632]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_01` Value : `navmeshes/v3_sc4_staging_01.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299666]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_02` Value : `navmeshes/v3_sc4_staging_02.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299699]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_03` Value : `navmeshes/v3_sc4_staging_03.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299738]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_04` Value : `navmeshes/v3_sc4_staging_04.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299781]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_05` Value : `navmeshes/v3_sc4_staging_05.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299821]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_06` Value : `navmeshes/v3_sc4_staging_06.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299862]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_07` Value : `navmeshes/v3_sc4_staging_07.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299902]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_08` Value : `navmeshes/v3_sc4_staging_08.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299942]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_09` Value : `navmeshes/v3_sc4_staging_09.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:299985]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_10` Value : `navmeshes/v3_sc4_staging_10.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:300024]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_11` Value : `navmeshes/v3_sc4_staging_11.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:300066]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_12` Value : `navmeshes/v3_sc4_staging_12.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:300110]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_13` Value : `navmeshes/v3_sc4_staging_13.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:300150]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_14` Value : `navmeshes/v3_sc4_staging_14.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:300188]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_15` Value : `navmeshes/v3_sc4_staging_15.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:300231]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_16` Value : `navmeshes/v3_sc4_staging_16.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[10:30:19:300272]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(305)::validateMap : `navmesh_instances` Key : `v3_sc4_staging_17` Value : `navmeshes/v3_sc4_staging_17.navmesh` not found on disk as absolute [... cut ...]
Bad inertia tensor properties, setting inertia to zero for link: r_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: l_gripper_finger_link
X Error of failed request:  BadAccess (attempt to access private resource denied)
  Major opcode of failed request:  150 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  87
  Current serial number in output stream:  87

QHCV, 2025-03-21:
I used the method below to solve this problem by modifying the .py file interactive-play.py.
[image]