facebookresearch/habitat-lab #2143: Error Occurs During Execution of python -m vlfm.run - Possible Semantic Annotation Issue?

Heartune, 2025-01-27:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.420230405

Habitat-Sim: v0.2.4

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/
Yes

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
Yes

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Dear Developers,

I followed the standard setup steps as documented, but encountered an error when running python -m vlfm.run. Below are the details of my process and the error:

Steps I Took:
```
conda_env_name=vlfm
mamba create -n $conda_env_name python=3.9 -y  
mamba activate $conda_env_name  
git clone https://github.com/bdaiinstitute/vlfm.git  
cd vlfm  
pip install torch==1.12.1+cu113 torchvision==0.13.1+cu113 -f https://download.pytorch.org/whl/torch_stable.html  
pip install git+https://github.com/IDEA-Research/GroundingDINO.git@eeba084341aaa454ce13cb32fa7fd9282fc73a67 salesforce-lavis==1.0.2  
pip install -e .[habitat]  
git clone https://github.com/WongKinYiu/yolov7.git  
./scripts/launch_vlm_servers.sh
```
Please note that I used mamba instead of conda due to the low speed of "solving environment...".

Error Observed:
After completing the above steps, I ran `python -m vlfm.run,` and the following error occurred:

> **_I’ve excerpted the most relevant parts of the error log, which might be lengthy. Thank you for your patience in reviewing this!_**

```
Script started on 2025-01-26 22:40:02+08:00 [TERM="xterm-256color" TTY="/dev/pts/1" COLUMNS="172" LINES="53"]

Traceback (most recent call last):
  File "/root/miniconda3/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/root/miniconda3/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/root/autodl-tmp/vlfm/vlfm/run.py", line 11, in <module>
    import frontier_exploration  # noqa
ModuleNotFoundError: No module named 'frontier_exploration'

[22:40:45:945365]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Dataset>: Proposing JSON name : default.scene_dataset_config.json from original name : default| This file does not exist.
[22:40:45:946031]:[Metadata] SceneDatasetAttributesManager.cpp(37)::createObject : File (default) not found, so new default dataset attributes created  and registered.
[22:40:45:946051]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Physics Manager>: Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json| This file does not exist.
[22:40:45:946062]:[Metadata] PhysicsAttributesManager.cpp(26)::createObject : File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
[22:40:45:946071]:[Metadata] MetadataMediator.cpp(210)::setActiveSceneDatasetName : Attempting to create new dataset ./data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json
[22:40:45:946088]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Dataset>: Proposing JSON name : ./data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json from original name : ./data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json| This file exists.

[22:40:45:946680]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/example/00861-GLAQ4DNUx5U/*.basis.glb
[22:40:45:946692]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/minival/00800-TEEsavR23oF/*.basis.glb
[22:40:45:946701]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/minival/00802-wcojb4TFT35/*.basis.glb
[22:40:45:946708]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/minival/00803-k1cupFYWXJ6/*.basis.glb
[22:40:45:946721]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/minival/00808-y9hTuugGdiq/*.basis.glb
[22:40:45:946734]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/test/00900-XRJ2muKkwAV/*.basis.glb
[22:40:45:946741]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/test/00901-aJoC8Qw6xQ5/*.basis.glb
[22:40:45:946751]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/test/00904-S9CUp5RsFY9/*.basis.glb
[22:40:45:946759]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/test/00905-9v5HzFAB1cm/*.basis.glb
[22:40:45:946769]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/test/00907-WGxjrSp71zX/*.basis.glb
[22:40:45:946778]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/test/00909-MrKys1cv4jD/*.basis.glb
[... 215 lines cut ...]
Exception ignored in: <function VectorEnv.__del__ at 0x7f078451a700>
Traceback (most recent call last):
  File "/root/.local/share/mamba/envs/vlfm/lib/python3.9/site-packages/habitat/core/vector_env.py", line 589, in __del__
    self.close()
  File "/root/.local/share/mamba/envs/vlfm/lib/python3.9/site-packages/habitat/core/vector_env.py", line 457, in close
    read_fn()
  File "/root/.local/share/mamba/envs/vlfm/lib/python3.9/site-packages/habitat/core/vector_env.py", line 108, in __call__
    res = self.read_fn()
  File "/root/.local/share/mamba/envs/vlfm/lib/python3.9/site-packages/habitat/utils/pickle5_multiprocessing.py", line 68, in recv
    buf = self.recv_bytes()
  File "/root/.local/share/mamba/envs/vlfm/lib/python3.9/multiprocessing/connection.py", line 216, in recv_bytes
    buf = self._recv_bytes(maxlength)
  File "/root/.local/share/mamba/envs/vlfm/lib/python3.9/multiprocessing/connection.py", line 414, in _recv_bytes
    buf = self._recv(4)
  File "/root/.local/share/mamba/envs/vlfm/lib/python3.9/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError: 
```

(vlfm) root@autodl-container-1b9e4e900f-1e494ad8:~/autodl-tmp/vlfm# exit

## Question:
I suspect this issue might be related to incomplete semantic annotations in the dataset or environment configuration. Could you kindly confirm if this error is caused by missing semantic annotations, or if there are additional steps I might have overlooked?

Thank you for your time and assistance!

closed 2025-02-09 by Heartune

Jay-Vim-Lv, 2025-02-11:
how you solved this problem?

Heartune, 2025-02-11:
> how you solved this problem?

我一开始用的是mamba来管理和配置虚拟环境，并手动下载数据集（因为用conda一直在Solving Environment）

后面我解决了Solving Environment的问题，所以用回了conda，数据集也是按照vlfm的github主页的说明来用python代码下载的，就没有这些问题了:)

Jay-Vim-Lv, 2025-02-11:
> > how you solved this problem?
> 
> 我一开始用的是mamba来管理和配置虚拟环境，并手动下载数据集（因为用conda一直在Solving Environment）
> 
> 后面我解决了Solving Environment的问题，所以用回了conda，数据集也是按照vlfm的github主页的说明来用python代码下载的，就没有这些问题了:)

Okay, thanks for your reply.