facebookresearch/habitat-lab #2152: SystemError: initialization of _internal failed without raising an exception

cirosantilli, 2025-02-17:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.3 or master?

Habitat-Sim: `habitat_sim==0.3.3` according to `pip freeze`

## 🐛 Bug

```
python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/
```

blows up with:

```
(habitat) ~/git/habitat-lab
2025-02-17_12-48-25@ciro@ciro-p14s$ python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/
Traceback (most recent call last):
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.3.3-py3.9-linux-x86_64.egg/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim import (
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.3.3-py3.9-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *  # noqa: F403
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.3.3-py3.9-linux-x86_64.egg/habitat_sim/agent/agent.py", line 12, in <module>
    import quaternion as qt
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/quaternion/__init__.py", line 31, in <module>
    from .quaternion_time_series import (
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/quaternion/quaternion_time_series.py", line 10, in <module>
    from quaternion.numba_wrapper import njit
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/quaternion/numba_wrapper.py", line 13, in <module>
    from numba import njit, jit, vectorize, int64, float64, complex128
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/numba/__init__.py", line 43, in <module>
    from numba.np.ufunc import (vectorize, guvectorize, threading_layer,
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/numba/np/ufunc/__init__.py", line 3, in <module>
    from numba.np.ufunc.decorators import Vectorize, GUVectorize, vectorize, guvectorize
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/numba/np/ufunc/decorators.py", line 3, in <module>
    from numba.np.ufunc import _internal
SystemError: initialization of _internal failed without raising an exception
(habitat) ~/git/habitat-lab
```

## Steps to Reproduce

Run installation steps from: https://github.com/facebookresearch/habitat-lab/tree/ba77964155815a6d6b04b4533ef0936a11a370d1?tab=readme-ov-file#installation until the offending command.

J1dan, 2025-02-20:
ChatGPT told me to `conda install --force-reinstall llvmlite numba -c conda-forge` and then it was solved

cirosantilli, 2025-02-20:
> ChatGPT told me to `conda install --force-reinstall llvmlite numba -c conda-forge` and then it was solved

Didn't work for me unfortunately, I still get the same error.

LatiosInAltoMare, 2025-02-22:
try this one:
```
conda remove llvmlite numba -y
conda install -c conda-forge numba=0.56.4
```
and then install `Habitat-sim` again using ```conda install habitat-sim withbullet -c conda-forge -c aihabitat ```

The problem should be solved. If `subprocess.CalledProcessError: Command '['git', 'lfs', 'pull']' returned non-zero exit status 1.` this occured, try this:

```
conda install -c conda-forge git-lfs
git lfs install
```
Then download the dataset using `conda install habitat-sim withbullet -c conda-forge -c aihabitat`

cirosantilli, 2025-02-25:
> try this one:
>
> ```
> conda remove llvmlite numba -y
> conda install -c conda-forge numba=0.56.4
> ```
>
> and then install `Habitat-sim` again using `conda install habitat-sim withbullet -c conda-forge -c [...]

OK, these appear to have moved things a bit forward!
 
> Then download the dataset using `conda install habitat-sim withbullet -c conda-forge -c aihabitat`

Did you mean to give a different command here? Because the original:

```
python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/
```

then blows up with:

```
Found the existing repo for (habitat_test_scenes): /home/ciro/git/habitat-lab/data/versioned_data/habitat_test_scenes
 checking out main and pulling changes from repo.
Traceback (most recent call last):
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.3.3-py3.9-linux-x86_64.egg/habitat_sim/utils/datasets_download.py", line 964, in <module>
    main(sys.argv[1:])
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.3.3-py3.9-linux-x86_64.egg/habitat_sim/utils/datasets_download.py", line 953, in main
    download_and_place(
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.3.3-py3.9-linux-x86_64.egg/habitat_sim/utils/datasets_download.py", line 755, in download_and_place
    checkout_repo_tag(repo, version_dir, version_tag)
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.3.3-py3.9-linux-x86_64.egg/habitat_sim/utils/datasets_download.py", line 632, in checkout_repo_tag
    repo.git.checkout(f"{tag}")
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/git/cmd.py", line 986, in <lambda>
    return lambda *args, **kwargs: self._call_process(name, *args, **kwargs)
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/git/cmd.py", line 1599, in _call_process
    return self.execute(call, **exec_kwargs)
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/git/cmd.py", line 1389, in execute
    raise GitCommandError(redacted_command, status, stderr_value, stdout_value)
git.exc.GitCommandError: Cmd('git') failed due to: exit code(1)
  cmdline: git checkout main
  stderr: 'error: The following untracked working tree files would be overwritten by checkout:
        .gitattributes
        README.md
Please move or remove them before you switch branches.
Aborting'
```

cirosantilli, 2025-02-25:
Oh I had missed the `git lfs install`. After that:

```
python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/
python -m habitat_sim.utils.datasets_download --uids habitat_test_pointnav_dataset --data-path data/
```

work at last!

But when I ran:

```
python examples/example.py
```

it failed with:

```
PluginManager::Manager: duplicate static plugin StbImageImporter, ignoring
PluginManager::Manager: duplicate static plugin GltfImporter, ignoring
PluginManager::Manager: duplicate static plugin BasisImporter, ignoring
PluginManager::Manager: duplicate static plugin AssimpImporter, ignoring
PluginManager::Manager: duplicate static plugin AnySceneImporter, ignoring
PluginManager::Manager: duplicate static plugin AnyImageImporter, ignoring
2025-02-25 18:30:57,752 Initializing dataset RearrangeDataset-v0
Traceback (most recent call last):
  File "/home/ciro/git/habitat-lab/examples/example.py", line 31, in <module>
    example()
  File "/home/ciro/git/habitat-lab/examples/example.py", line 15, in example
    with gym.make("HabitatRenderPick-v0") as env:
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/gym/envs/registration.py", line 676, in make
    return registry.make(id, **kwargs)
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/gym/envs/registration.py", line 520, in make
    return spec.make(**kwargs)
  File "/home/ciro/miniconda3/envs/habitat/lib/python3.9/site-packages/gym/envs/registration.py", line 140, in make
    env = cls(**_kwargs)
  File "/home/ciro/git/habitat-lab/habitat-lab/habitat/gym/gym_definitions.py", line 91, in _make_habitat_gym_env
    env = make_gym_from_config(config)
  File "/home/ciro/git/habitat-lab/habitat-lab/habitat/gym/gym_definitions.py", line 60, in make_gym_from_config
    return make_env_fn(env_class=env_class, config=config, dataset=dataset)
  File "/home/ciro/git/habitat-lab/habitat-lab/habitat/utils/env_utils.py", line 36, in make_env_fn
    dataset = make_dataset(config.dataset.type, config=config.dataset)
  File "/home/ciro/git/habitat-lab/habitat-lab/habitat/datasets/registration.py", line 22, in make_dataset
    return _dataset(**kwargs)  # type: ignore
  File "/home/ciro/git/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py", line 59, in __init__
    raise ValueError(
ValueError: Requested RearrangeDataset config paths 'data/datasets/replica_cad/rearrange/v2/train/rearrange_easy.json.gz' or 'data/replica_cad/' are not downloaded locally. Aborting.
```

so presumably there's some other download command missing. Any clues whats the missing download command?

cirosantilli, 2025-02-27:
There's a:

```
python -m habitat_sim.utils.datasets_download --uids replica_cad_dataset --data-path
```

and it appears to produce:

```
data/replica_cad
```

but the issue persists.

RayDebjit, 2025-03-20:
> Oh I had missed the `git lfs install`. After that:
>
> ```
> python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/
> python -m habitat_sim.utils.datasets_download [...]