facebookresearch/habitat-lab #2181: Gibson dataset v1.1 in objectnav task cannot used

Benson722, 2025-05-08:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.3

Habitat-Sim: v0.3.3

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/
Yes
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
Yes
Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
Hi!
I want to use gibson dataset in objectnav task. The yaml configs I use are below:

1. dependency/habitat-lab/habitat-lab/habitat/config/habitat/dataset/pointnav/gibson.yaml:
```
# @package habitat.dataset

defaults:
  - /habitat/dataset: dataset_config_schema
  - _self_

type: PointNav-v1
split: train
# data_path: data/datasets/pointnav/gibson/v1/{split}/{split}.json.gz
data_path: data/datasets/objectnav/gibson/v1.1/{split}/{split}.json.gz
```

2. dependency/habitat-lab/habitat-lab/habitat/config/habitat/task/objectnav.yaml (No edited by me)
3. dependency/habitat-lab/habitat-lab/habitat/config/benchmark/nav/objectnav/objectnav_gibson_with_semantic.yaml:
```
# @package _global_

defaults:
  - /habitat: habitat_config_base
  - /habitat/task: objectnav
  - /habitat/simulator/sensor_setups@habitat.simulator.agents.main_agent: rgbd_semantic_agent
  - /habitat/dataset/pointnav: gibson
  - _self_

habitat:
  environment:
    max_episode_steps: 500
  simulator:
    turn_angle: 30
    agents:
      main_agent:
        sim_sensors:
          rgb_sensor:
            width: 640
            height: 480
            hfov: 79
            position: [0, 0.88, 0]
          depth_sensor:
            width: 640
            height: 480
            hfov: 79
            min_depth: 0.5
            max_depth: 5.0
            position: [0, 0.88, 0]        
          semantic_sensor:
            width: 640
            height: 480
            hfov: 79
[... 161 lines cut ...]
            │   │   ├── Ranchester.glb.json.gz
            │   │   ├── Wainscott.glb.json.gz
            │   │   └── Woodbine.glb.json.gz
            │   ├── train_info.pbz2
            │   └── train.json.gz
            └── val
                ├── content
                │   ├── Collierville_episodes.json.gz
                │   ├── Collierville.glb.json.gz
                │   ├── Corozal_episodes.json.gz
                │   ├── Corozal.glb.json.gz
                │   ├── Darden_episodes.json.gz
                │   ├── Darden.glb.json.gz
                │   ├── Markleeville_episodes.json.gz
                │   ├── Markleeville.glb.json.gz
                │   ├── Wiconisco_episodes.json.gz
                │   └── Wiconisco.glb.json.gz
                ├── val_info.pbz2
                └── val.json.gz

7 directories, 36 files
```

The content of "train.json.gz" and "val.json.gz" is:
`{"episodes": []}`

Benson722, 2025-05-31:
> 这些数据都是哪来的呀，好像没有gibson对应的objectnav数据集吧（

It is from [PONI](https://github.com/srama2512/PONI). :(

labixiaochao, 2025-09-02:
> > 这些数据都是哪来的呀，好像没有gibson对应的objectnav数据集吧（
> 
> It is from [PONI](https://github.com/srama2512/PONI). :(

Have you been successful in downloading and creating Gibson and eval PONI? I also encountered some problems, can we add a contact information to communicate?